worlickwerx/pi-parport #2: implement IRQ on low to high nACK transition

garlick, 2017-08-06:
The optional parallel port interrupt is supposed to be generated on a low to high transition on pin 10 (nACK).

http://www.makelinux.net/ldd3/chp-10-sect-1

http://retired.beyondlogic.org/spp/parallel.htm

See `gpiod_to_irq()`
https://www.kernel.org/doc/Documentation/gpio/consumer.txt

andersmilton, 2019-09-09:
Do you have any plans to incorporate interrupts in the driver? I have a project in mind that surely could use it, but I don't know anything about writing linux drivers :)

garlick, 2019-09-09:
I don't think it will be that difficult.  The main thing holding me back was having a compelling use case and some plan for testing.  What's your project if you don't mind me asking?

andersmilton, 2019-09-09:
I'm thinking of building a Pi Zero hat version of the XA1541 cable for communicating with old Commodore disk drives :D

I'm literally just thinking about it at the moment. I'm not sure it will ever be built, but it should be easy hardware-wise, and the software exists.

quorten, 2020-11-01:
A suitable test case for this is rather easy, the `lp` driver for printing via parallel port requires nACK to know when to send the next character.  Without interrupts, it resorts to polling.  This works fine for this particular driver, but I believe other for device drivers like cheat cartridges it may cause issues.

So, I think I might get into adding this feature and testing it.

garlick, 2024-08-29:
Oh this might be something we ought to add to the v5 board proposed in #68.  I've kind of flushed my cache on how parallel port interrupts supposed to work.   A little study required on my part I guess.

garlick, 2024-08-29:
Actually, I guess apart from making sure that nACK is assigned to a GPIO pin that can trigger an interrupt, this is a software problem not a hardware one.  It's GPIO27 in v5 and GPIO17 in older versions.  AFAICT all pi gpio pins are capable of generating an interrupt on either low to high or high to transition.

So never mind.  No hardware support needed.