worlickwerx/pi-parport #13: Possible to do a rpi b (rev2.0) hat?

orrpan, 2018-08-08:
If soldering additional pins (p5-pins) would it be possible to use an older rpi, only by changing GPIO ports used and the schematics of the pcb?

http://pi4j.com/pins/model-b-rev2.html

garlick, 2018-08-14:
Yeah, maybe!  Although given that there are several generations of pi now that use the modern GPIO pinout, is it worth the effort?   Then again, most people would say this whole project isn't worth the effort and we should let the parallel port die its natural death, so who am I to talk :-)   Best of luck.

garlick, 2020-01-02:
Closing as this is probably not worth the effort given that pi foundation has more or less standardized the 40 pin connector for a while now.

closed 2020-01-02 by garlick