worlickwerx/pi-parport #14: document process for compiling driver against pi kernel

garlick, 2018-09-01:
It's non-obvious how to compile the kernel driver, so a little documentation could be helpful.

Some notes, working backwards since I didn't take any when I worked through this before:

* Install kernel headers with `sudo apt-get install raspberrypi-kernel-headers`
* The driver Makefile references `/lib/modules/$(KERNEL_VERSION)/build` where KERNEL_VERSION is taken from `uname -r`.
* If not running the same kernel as the one you are building for, you can run `make KERNEL_VERSION=whatever`
* Parport stack is configured off by default in the raspberry pi kernel, so it is not built
* This parport stack code was copied into driver directory in this project
* Ideally, driver would be submitted upstream, pi maintainers convinced to enable parport stack, and this duplication of code eliminated; meanwhile, copied sources may need to be synchronized from time to time when upstream changes
* You have to consistently use the packaged kernel and headers for this to continue to work through updates;  don't pull in bleeding edge kernel with  `rpi-update` or this procedure won't work.

garlick, 2018-09-01:
Output from running `make` in the driver directory should look similar to this:
```
garlick@sbig9e:~/proj/pi-parport/driver $ make clean
make -C /lib/modules/4.14.62-v7+/build M=/home/garlick/proj/pi-parport/driver clean
make[1]: Entering directory '/usr/src/linux-headers-4.14.62-v7+'
  CLEAN   /home/garlick/proj/pi-parport/driver/.tmp_versions
  CLEAN   /home/garlick/proj/pi-parport/driver/Module.symvers
make[1]: Leaving directory '/usr/src/linux-headers-4.14.62-v7+'
garlick@sbig9e:~/proj/pi-parport/driver $ make
make -C /lib/modules/4.14.62-v7+/build M=/home/garlick/proj/pi-parport/driver modules
make[1]: Entering directory '/usr/src/linux-headers-4.14.62-v7+'
  CC [M]  /home/garlick/proj/pi-parport/driver/share.o
  CC [M]  /home/garlick/proj/pi-parport/driver/ieee1284.o
  CC [M]  /home/garlick/proj/pi-parport/driver/ieee1284_ops.o
  CC [M]  /home/garlick/proj/pi-parport/driver/procfs.o
  CC [M]  /home/garlick/proj/pi-parport/driver/daisy.o
  CC [M]  /home/garlick/proj/pi-parport/driver/probe.o
  LD [M]  /home/garlick/proj/pi-parport/driver/parport.o
  CC [M]  /home/garlick/proj/pi-parport/driver/parport_gpio.o
  CC [M]  /home/garlick/proj/pi-parport/driver/lp.o
  CC [M]  /home/garlick/proj/pi-parport/driver/ppdev.o
  Building modules, stage 2.
  MODPOST 4 modules
  CC      /home/garlick/proj/pi-parport/driver/lp.mod.o
  LD [M]  /home/garlick/proj/pi-parport/driver/lp.ko
  CC      /home/garlick/proj/pi-parport/driver/parport.mod.o
  LD [M]  /home/garlick/proj/pi-parport/driver/parport.ko
  CC      /home/garlick/proj/pi-parport/driver/parport_gpio.mod.o
  LD [M]  /home/garlick/proj/pi-parport/driver/parport_gpio.ko
  CC      /home/garlick/proj/pi-parport/driver/ppdev.mod.o
  LD [M]  /home/garlick/proj/pi-parport/driver/ppdev.ko
make[1]: Leaving directory '/usr/src/linux-headers-4.14.62-v7+'
```

andrewtubbiolo, 2019-06-17:
Have you tried 4.19.42-v7+ yet? The /lib/modules/4.19.42-v7+/build directory is devoid of any files at all.
I see you've had success with 4.14.62-v7+. Should I build a system that uses that lib version?

garlick, 2019-06-17:
No I haven't.  By chance did you update with `rpi-update`?

If you use the normal apt update process, then `raspberrypi-kernel-headers` provides `/usr/src/linux-headers-${version}` which `build` is a symlink to.

I'm not sure if there is a documented process to get headers that match a kernel installed with `rpi-update`... 

https://www.raspberrypi.org/documentation/linux/kernel/updating.md

garlick, 2019-06-17:
Just pulled down the latest stretch updates with apt, and that kernel was installed.  My `build` dir is populated.  Make sure you have `raspberrypi-kernel-headers` installed.

andrewtubbiolo, 2019-06-17:
I did not use rpi-update. I did read your docs. :) It's been a few years (overa decade) since I've compiled a linux kernel, so I'm going back up the curve with this.

I applied the symlink by hand and with some success, but got these errors.

sudo ln -s /usr/src/linux-headers-4.19.42-v7+ /lib/modules/4.19.42-v7+/build

/home/pi/src/pi-parport-master/driver/parport/ieee1284.c:80:2: error: implicit declaration of function 'init_timer_on_stack' [-Werror=implicit-function-declaration]
  init_timer_on_stack(&timer);
  ^~~~~~~~~~~~~~~~~~~
/home/pi/src/pi-parport-master/driver/parport/ieee1284.c:82:17: error: assignment from incompatible pointer type [-Werror=incompatible-pointer-types]
  timer.function = timeout_waiting_on_port;
                 ^
/home/pi/src/pi-parport-master/driver/parport/ieee1284.c:84:7: error: 'struct timer_list' has no member named 'data'
  timer.data = port->number;

So it seems I symlinked to the wrong directory, or indeed some things have changed.

andrewtubbiolo, 2019-06-17:
Well bugger, yes, I have raspberrypi-kernel-headers, undid the symlink, did a dist-upgrade and the build directory is still empty. Okay, I'll start over with a clean sheet and see if that sets things right. 

Thanks.
Andrew

andrewtubbiolo, 2019-06-17:
A clean start, with normal raspi-config session, an apt-get install of raspberrypi-kernel-headers, git, a clone of this project, made for a clean compile with kernel mods showing on an lsmod. Logic analyser sessions tomorrow. 

Thanks.
Andrew

garlick, 2020-01-02:
I tried to clarify this process in the README.md.  Hopefully it's better.

closed 2020-01-02 by garlick