worlickwerx/pi-parport #18: Pi Zero Board (With DB25)

danhans42, 2018-11-01:
H,

Just wanted to say thank you for this project, I have seen many a request for such a device over the years and I am surprised this has not been more widely used. 

I am currently using this on a Pi Zero (or will be), but have had to opt for a TSSOP parts and a breakout board to get a demo board up and running.

I just wondered if you would consider at some rerouting your board so it fits nicely on a Pi Zero with a PCB mount DB25 connector rather than the DIL? I have loaded your files into Eagle but is currently way beyond me.

If not, no problem - think I had better get learing Eagle :-)

Thanks for your work so far!

garlick, 2018-11-01:
Hi, glad you're having fun with it (or at least planning for future fun :-)

You might try Arrow Electronics for the SSOP parts.

Or doing a new board that fits the Pi Zero and uses a TSSOP part might be a good idea, though I don't presently have time to work on it.  (If you got one working I'd certainly point to your work from here, or pull it in if you wanted!) The driver of course should "just work" with the new board - you could even use different GPIO pins convenient for layout as long as a separate device tree overlay file was provided for that board.

I've been planning to switch to KiCAD for a while.  Eagle's subscription licensing kind of sucks, so I'm stuck on the older version before they changed the licensing.  If I was starting without any Eagle knowledge, I'd surely go with KiCAD, since Eagle has many oddities.  (My humble opinion)

garlick, 2018-11-01:
One thing - my first prototype had a DB25 on the board, and I had some problems with mechanical stress on the board, plus I found it to be a PITA to accommodate the connector in compact off-the-shelf pi case.  Admittedly the board was a crappy home-made job, without plated through holes.

danhans42, 2018-11-01:
Thanks for getting back to me.

Basically my tests will be using a cheat cartridge for a game console. So by using a PCB mounted connector I was hoping to plug the whole unit into the device itself (it's the xplorer device for the Sony play station), so no cable would be required.

I will have a look at options for designing a PCB, the whole world of sending a design is fairly new to me - the last PCB I designed was single sided and printed using my old HP Laserjet 4 using ExpressPCB.

My thoughts of using the Pi Zero is so it's a small compact unit that just plugs straight into my device.

I will take a look at learning Eagle or starting from scratch as I also wanted to break out the UART pins to a header.

I will start taking a look next week so long as my test setup works correctly.

Thanks for the pointers so far, most helpful.

garlick, 2018-11-01:
Oh interesting - another guy used this board (successfully) for a legacy game cartridge reader/writer.  He was using software called ucon64, which could be configured to use the "ppdev" user-space interface to the parport.  That aspect is critical to being successful with this board.  If the software tries to access the legacy ports by address, it won't work.

danhans42, 2018-11-02:
Definitely works with ppdev, the software is here

danhans42, 2018-11-04:
Just an update, I have this working pretty well in a test setup.. I have used some crude level shifting methods and managed to omit both strobe and status (the cheat carts dont use them).

For what I need to do I might just have a look at using something along the lines of your V1 schematic as this is pretty much what I followed (I just omitted the '74 series stuff for resistor dividers - not good).

Now I know it definitely works I will look at designing a small board for it to go on.

Just to say, thanks again! This is something that I have wanted to achieve for years - the cheap RaspberryPi Zero and your great work has made this possible.

Regards

Dan

garlick, 2018-11-04:
Nice!  That was fast.

garlick, 2020-01-02:
FYI: this project was reworked in KiCAD so it should be easier to lay out a new board for the Pi Zero form factor if someone feels the need.

quorten, 2020-07-27:
F.Y.I. I started a draft to rework this to Pi Zero form factor.  Sneak peak at first iteration as a HAT.

[image]

RPi Zero WH would probably be ideal for this form.  However, with other Raspberry Pi Zero versions that do not have the pin header soldered on, there is a better option: use two right-angle pin header connectors to put the HAT board on the same horizontal level as the Zero board, then a DB25 connector is board-mounted on the opposite end of the HAT.  This allows the whole assembly to be put in a nice square case that is easy to get a hand grip on when working with DB25 connectors that are often stiff to plug/unplug, and it positions the USB & HDMI connectors on the opposite end of the DB25 connector.

That being said, it might make sense to make two versions of the board for RPi Zero.

garlick, 2020-07-28:
Nice!

Also, funny, I just oshparked a 40 pin adapter that does exactly that.  Two versions, one for regular tall headers, and one surface mount for low profile.  I'll share them with you if you're interested, not that it's rocket science :-)

garlick, 2020-07-28:
[this one](https://oshpark.com/shared_projects/KMQXVn0e) should work with Adafruit's $2 [low profile socket](https://www.adafruit.com/product/2187)

[this one](https://oshpark.com/shared_projects/2UQIWw9e) works with std through hole

Design files  https://github.com/garlick/pi-cluster-two/tree/master/hardware/40pin-adapter