worlickwerx/pi-parport #19: DMA

masterzion, 2019-02-26:
Hi, 
I have a doubt about this lib. I am not sure if it is a issue.

I after build and run, it seems it works.
I tested whit this Diagnostic Utility.
http://yyao.ca/projects/ParallelPortLinux/

But, it seems I must to declare the DMA address to make it work with Linux CNC
[image]

I could not find the address on the documentation and it seems the 0x378 is not recognized by hal_parport.

Is it possible to set some address?
The default linux parport_pc allow us to set it by parameter.
ex:
insmod parport_pc io=0x3b

Regards

garlick, 2019-02-26:
This board doesn't support dma.  I am very interested to hear how you get
on with linuxcnc though!  I hope dma is optional?

garlick, 2019-04-25:
There was a way to tell the PC DMA controller to move data through the parallel port in ECP mode, but looking at your issue in more detail, I think linuxcnc just wants to bit bang the legacy x86 memory mapped I/O ports.

That won't work here.  If one wants to bit bang the port pins, one would need to go through the parport driver stack using ppdev, described here:

http://as6edriver.sourceforge.net/Parallel-Port-Programming-HOWTO/accessing.html

I don't know if the additional latency of the ppdev method is prohibitive for this use case, but since apparently there is a hardware abstraction layer (HAL?) it might be reasonable to try to create a driver that uses ppdev, or perhaps one that allows you to more directly access arbitrary GPIO pins?

I'll retitle this issue to be more specific to linuxcnc, since it's a compelling use case if it could work!

andrewtubbiolo, 2019-04-25:
I use LinuxCNC on my milling machines! When I was thinking of doing TDI
with a Pi, my first thought was LinuxCNC with it's real time capability
would probably be the solution. But I thought an FPGA would be easier.

Andrew

Dloranger, 2020-12-10:
I am interested in using this for linux CNC as well, has anyone gotten this to work?

garlick, 2020-12-10:
Not that I'm aware of.

vonj, 2020-12-10:
Sounds plausible that it should work, actually.

vspositron, 2020-12-29:
Hi Everyone,

I am eagerly waiting for an RPI hat that will enable me to connect with the cheap LinuxCNC/ Mach3 CNC boards available in the market (https://in.pinterest.com/pin/720013059148220267/). It will be a holy grail for the DIY community. Please include this capability in the next version. I also appreciate all the efforts you and the team have taken so far on this project.

Thanks,
Sagar

garlick, 2020-12-29:
Maybe it could work now if, instead of using ppdev + the kernel parport stack,  [libbcm2835](https://www.airspayce.com/mikem/bcm2835/) is used to memory map the Pi's GPIO peripheral where it can be directly poked from user space.

That would be kind of like directly accessing the legacy parallel port on an x86 using its I/O address, which apparently works for linuxcnc.

I don't follow linuxcnc, but maybe someone has already tried this approach?

garlick, 2020-12-29:
Heh, just found this:

https://github.com/LinuxCNC/linuxcnc/blob/master/src/hal/drivers/hal_pi_gpio.c

garlick, 2020-12-31:
There is also this:

https://github.com/LinuxCNC/linuxcnc/blob/master/src/hal/drivers/hal_pi_gpio_demo.hal

which shows that the gpio pin assignment is configurable.  Hmm, I don't know if the hal driver for the parport does any other magic to make those parallel port stepper boards work, but it _may_ be as simple as replicating the above example with the correct list of GPIO pin numbers for our board + parallel port stepper board?