worlickwerx/pi-parport #23: clarify hardware license

garlick, 2019-08-04:
The hardware portion of this project was intended to be freely shared and extended.

The project LICENSE is GPL-2.0.  The hardware design I guess would default to GPL-2.0, but that may not be an appropriate license for hardware.  Investigate the state of the practice for "open source hardware" and adopt one of the best practices in this project. 

See: https://www.oshwa.org/definition/

Consider porting the design files from Eagle (proprietary) to KiCAD (free), and silk screening the OSHWA logo on the next version of the board.

andersmilton, 2019-09-09:
I don't really like GPL for hardware myself. I usually go for CC BY-SA for hardware and GPL for software. But GPL should work.

garlick, 2019-09-09:
Yeah that might be the way to go.

garlick, 2020-01-02:
Went with CERN-OHL-1.2

closed 2020-01-02 by garlick