worlickwerx/pi-parport #30: [v3] generate BOM

garlick, 2020-01-03:
Need to generate bill of materials for version 3

quorten, 2020-03-11:
Thanks for making pi-parport, and thanks for reworking the hardware in KiCad!

I was looking around at this and thought I might add some improvements while I'm at it.  One improvement that I'm thinking about for the V3 Bill of Materials is specific manufacturer part number references for all parts rather than just the ICs.  Also, large electronics distributors like Digi-Key may also have their own part numbers.  These can be added into the schematic and the Bill of Materials can be generated from KiCad.  Then, that can be used to easily place an order for all parts from a single distributor.

I'll probably have a pull request coming your way soon with my ideas.

garlick, 2020-03-12:
Awesome!  Yep, I think the idea is you would enter that stuff as device metadata that the BOM generator in kicad can pull from.  There's a "John's basement" episode on it somewhere in this playlist:

https://www.youtube.com/playlist?list=PL3by7evD3F51fKkyrUbH-PCdwPCWc9F8a

garlick, 2020-03-13:
Closed by #32

closed 2020-03-13 by garlick