worlickwerx/pi-parport #35: hardware/v4: build and test

garlick, 2020-07-28:
@quorten - did you have any other changes in mind for v4 of should we consider it stable?

I was thinking of ordering boards, building and testing one prototype, and publishing the design on oshpark.

quorten, 2020-07-28:
@garlick I was thinking v4 should probably focus on fixing minor issues, and adding features (and more issues) would be kept to later versions.  So yeah, I'd call it stable.

Mentioning building one version, I saw Digi-Key is currently out-of-stock on the SN74LVC161284DLR chip, expected shipment in mid September.  (They usually over-estimate the lead time.)  Arrow is still in stock though.  Tactfully said, apparently building parallel port interfaces is all the rage these days.

By the way, if OSHPARK has a minimum quantity of 3 and you don't need all 3 of the boards, I'd be more than happy to get one of the extra boards, if you don't mind.

garlick, 2020-07-28:
You bet!  Send me your shipping address and I'll send you a board and a few chips.  garlick dot jim at gmail dot com.

quorten, 2020-10-22:
Finally got around to testing my built PCB, and it is a success!  Printing images to an Epson Stylus Color 640 printer works great.  Unfortunately I was unable to test high-speed data input because my old parallel port scanner never had a Linux driver written due to missing protocol documentation.

@garlick If you think this is sufficient testing on hardware v4, you can probably close this issue now.

garlick, 2020-10-22:
Great to hear!  Sure let's close.   BTW, just merged #37 to work with latest raspbian kernel.  If you have a chance to update and retest with latest master, please open an issue if you have any problems.  I just compile tested it and checked that the target kernel was the latest raspbian.

closed 2020-10-22 by garlick