worlickwerx/pi-parport #38: travis-ci: re-enable

garlick, 2020-10-21:
Travis is not running on pull requests.

Also, we should add 5.4.51 to the matrix as the GPIO API changed.

Add conditional code for earlier kernels too, as #37 will have broken builds for those.

garlick, 2020-12-31:
At this point, travis seems to no longer be the CI of choice for open source projects, so changing title.

Geato, 2020-12-31:
Hi everyone.

I have been watching this project from a distance purely from the standpoint of hoping it eventually works for Linuxcnc. I am not a coder but I have many years experience with electronic systems design. I am willing to offer any assistance if I can.

The best way to assemble this board is to use a solder paste screen and "silk" screen the solder paste on the board, place the components then place the board on a hotplate for a few minutes until the solder flows. The through hole components can be hand soldered afterwards.

Unfortunately you do not want to have to order a screen for every board ordered. The screen is 10$ but one must also purchase the solder paste which is probably 2-3 times the cost of the screen.

The best scenario in terms of cost is someone supplies partially assembled kits with the smt components already mounted and let the customer finish the through hole components since that is where the labor costs are.

Just my 2 bits worth.

garlick, 2020-12-31:
Thanks for your comments @Geato ! I think you meant this for #48 ?

Also did you see issue #19 on linuxcnc?

commit 6f8a7d18 (2021-01-07): ci: drop travis config

Problem: Travis is no longer offering the same CI service
to open source projects that it once did, and most projects
have migrated to github workflows.

Drop travis config now that we have a rudimentary github
workflow set up.

Fixes #38
closed 2021-01-07 by quorten