worlickwerx/pi-parport #46: Cannot compile Device Tree Overlay

afromanDAN, 2020-12-26:
I'm trying to compile the Device Tree Overlay on a raspberry pi 0 and I get this error:

`pi@raspberrypi:~/pi-parport/dts $ make`
`dtc -@ -I dts -O dtb -o parport-gpio.dtbo parport-gpio.dts`
`parport-gpio.dtbo: Warning (unit_address_vs_reg): /fragment@1/__overlay__/ppgpio@0: node has a unit name, but no reg property`

garlick, 2020-12-26:
Please check, but the dtbo file should have been produced as it's a non fatal error.

Here's a forum discussion that suggests these warnings are to be expected:

https://www.raspberrypi.org/forums/viewtopic.php?t=172342

specifically the post from Phil where he points out:

> `Node /fragment@0 has a unit name, but no reg property`
> Here, dtc is tricked by the fragment@<n> syntax into expecting a corresponding "reg", but there shouldn't be one. This warning is a fact of life until dtc is updated to include a better understanding of the overlay syntax.

afromanDAN, 2020-12-27:
Ah, ok. Thanks for the explanation.

closed 2020-12-27 by afromanDAN