worlickwerx/pi-parport #47: R1 missing from the v4 BOM

afromanDAN, 2020-12-29:
What part is it on the board? I built the board but noticed it was missing.

quorten, 2020-12-29:
This is labeled as a "DNP" (Do Not Place) part, the purpose is if you
want to hard-wire write-protect rather than using the removable
jumper.  A 220 ohm resistor or less could work just fine, but this
part is not necessary for the board.

quorten, 2020-12-29:
Out of curiosity, how did you print the PCB?  And which assembly
technique did you use, reflow soldering or hand soldering + drag
soldering for the LPT chip?

I have a more detailed assembly guide with photos from my second board
assembly, but I haven't put it online yet.  But point in hand, it's
worth determining what info would be most helpful to relative novices
in this area.  And also I'm curious about the relative interest in
this project, I'm guessing hand-assembling a small number of PCBs
suits this project just fine but there could be the option of PCB
assemblers for readymade boards in larger quantity.

afromanDAN, 2020-12-29:
Well I tried reflow soldering on one of them but my solder paste applicator put too much on. Second one I did was with just hand soldering but i haven't tested it yet because I'm waiting on a terminator. I think the easiest way to make one of these is to use a solder paste template and a hot air gun.

andrewtubbiolo, 2020-12-29:
Hey Dan:

     I got one of those $40 hot air units from China on Amazon. It did a
great job. I mounted mine on a surface
mount to DIP 40 converter then hand soldered wires. I simple thin line
across the pads and surface tension
did the rest. I hand tacked the corners with some angle hair solder and a
small chisel iron tip. I had two
or three bridges I cleaned up with some solder braid.

   I'm going to be doing a rev 4 after I order the board next week.

Andrew

quorten, 2020-12-30:
Cool, my guide is largely photographed and written around a solder paste technique without templates.  Granted my second board is not quite finished yet, but sounds like my recommendations are largely on track with what works well for other folks.  I think I may post the detailed info I currently have in the next few days.

afromanDAN, 2020-12-30:
Well I think I cooked the lpt chip. I would appreciate it if someone could make one for me.

quorten, 2020-12-31:
@afromanDAN What makes you think that?  Did you power it on and it didn't work?  Multimeter tests to check for soldering problems on powered-off ICs can fool you because the 3.3v side has ESD protection diodes, so it should appear that all such pins are "bridged" together from an electrical test.  But when you power on, it will "just work" so long as you've visually verified there are no bridges.

I'd be willing to mail you out my one finished and tested board if you're interested.

@garlick It might be worth creating a GitHub issue to discuss more about the possibility of stocking partially/fully assembled v4 boards.  Also, thanks for sending me the two PCBs!

garlick, 2020-12-31:
Thanks for helping out @quorten!  I'll open an issue.

afromanDAN, 2021-01-01:
@quorten That would be great if you could ship one out to me.

I'll close this issue now.

closed 2021-01-01 by afromanDAN

quorten, 2021-01-01:
@afromanDAN Send me an E-mail with your address and we can follow up with additional details there.  Same as my GitHub E-mail address.  mako0042 at umn dot edu.

quorten, 2021-01-05:
@garlick Digging deeper into this, there's a few things we should amend to our documentation and one minor bug in my driver install code that should be fixed.  Good news is that one of the two boards @afromanDAN built actually fully works!  I'll open another pull request soon.