worlickwerx/pi-parport #48: explore possible kit offering

garlick, 2020-12-31:
There are some barriers for people getting this project off the ground:
- fine pitch SMT soldering
- high shipping to part cost for small number of components
- single quantity part prices are high
- oshpark PCB fab cost is somewhat high for this size board and you have to buy 3

garlick, 2020-12-31:
A test fixture might be nice, to check out a board that has only the SMT parts soldered.

It could
* hold the board with pogo pins
* power the board
* exercise the I/O lines
* program the EEPROM

Or am I getting off in the weeds for a demand that may add up to 5 boards?  :-)

BTW, PCBWAY estimates $60 for 40 boards including international shipping.

quorten, 2020-12-31:
I think a test fixture is getting into the weeds for SMT-only soldered boards.  For other small-volume projects, it seems sane to ship out factory-soldered boards as-is and leave testing up to the user who populates on the through-hole components to complete it.

quorten, 2020-12-31:
Oh yeah... pi-parport is much cheaper in volume than RaSCSI because it doesn't need relatively rare and expensive transceiver chips.  It could just as well make sense to only offer fully assembled boards.

garlick, 2020-12-31:
Maybe we should put the BOM into digikey and see where the price breaks are? (Probably dominated by the price break on the buffer chip which is at  qty 10, 25, and 100)

At minimum it seems like a no brainer to put together a parts kit that could be provided to people at cost + shipping, since it will be so much cheaper for someone that wants to assemble one or two boards.

quorten, 2020-12-31:
The 26-pin header, 40-pin header, and buffer chip together have the least discounting at larger volume and are about $3-$4 total.

For the BOM quantities as stated (i.e. not including spares for small passives), quantity 10-50 is largely similar price breaks.  So, price breaks for 5 boards, 10 boards, and 100 boards, these are the totals for the BOM of one board.

* Qty 5: $7.395
* Qty 10: $5.518
* Qty 100: $4.4055

garlick, 2020-12-31:
Thanks for doing that!

To add to the data, here is the cost per board from PCBWAY (shipped) with a stencil thrown in:
* Qty 5: $11.60
* Qty 10: $6.40
* Qty 20: $4.20
* Qty 100: $1.36

Putting together a kit with the SMT parts soldered on as suggested by @Geato in #38 might not be so bad.

quorten, 2021-01-02:
Including the DIP jumper is probably a good idea for the kit offering, I'll update the BOMs to include the Digi-Key PN of the DIP jumper I've been using.

quorten, 2021-01-06:
It's probably a good idea to perhaps set something up to gauge interest from folks who might be interested in pre-ordering a kit.  Could be as simple as a note in the top-level README.md for adding a comment to this GitHub issue.

But yeah, my hunch is that an initial batch of 5 boards could last at least a year, 10 boards twice as long.  But I'm basing this off of my (possibly false) assumption that the pi-parport market has already been relatively "saturated" by those who assembled their own boards.

quorten, 2024-08-18:
I've been thinking about this for a long while and will jot down my thoughts here.

To avoid sitting on too much inventory, we can think of things this way: we are providing kits first as a convenient source of spares for current users, and second as an easy way for new users to get started.

Starting out by making a batch of 10 kits seems to be a reasonable ask.  That way, the total up-front cost for one batch at a time is somewhere between $120 and $300.  I don't know the current prices, but it seems like a manageable ballpark.

quorten, 2024-08-20:
Sure, I have time to review some changes.