worlickwerx/pi-parport #50: does a PCIe parallel port card work with the CM4?

garlick, 2021-01-02:
@herrkuhn mentioned in #31 that he had tried a parallel port PCIe card with the Pi CM4 (I assume with the CM4 IO module which brings the single PCIe lane out to a x4 slot).

Let's capture any data from that experiment here for people who might be looking at other ways to get a parallel port to work with the Pi.

herrkuhn, 2021-01-02:
Here's what I get with a Raspberry Pi 4/CM4 and a Delock PCI Express Card to 1 x Parallel IEEE1284 (https://www.delock.de/produkt/90412/pdf.html?sprache=en) in the PCIe port:

```
pi@raspberrypi:~ $ sudo lspci -vvv
00:00.0 PCI bridge: Broadcom Limited Device 2711 (rev 20) (prog-if 00 [Normal decode])
	Control: I/O- Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr+ Stepping- SERR+ FastB2B- DisINTx-
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Latency: 0, Cache Line Size: 64 bytes
	Interrupt: pin A routed to IRQ 55
	Bus: primary=00, secondary=01, subordinate=01, sec-latency=0
	I/O behind bridge: 00000000-00000fff
	Memory behind bridge: f8000000-f80fffff
	Prefetchable memory behind bridge: 00000000fff00000-00000000000fffff
	Secondary status: 66MHz- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- <SERR- <PERR-
	BridgeCtl: Parity+ SERR+ NoISA- VGA- MAbort- >Reset- FastB2B-
		PriDiscTmr- SecDiscTmr- DiscTmrStat- DiscTmrSERREn-
	Capabilities: [48] Power Management version 3
		Flags: PMEClk- DSI- D1- D2- AuxCurrent=0mA PME(D0+,D1-,D2-,D3hot+,D3cold-)
		Status: D0 NoSoftRst+ PME-Enable- DSel=0 DScale=1 PME-
	Capabilities: [ac] Express (v2) Root Port (Slot-), MSI 00
		DevCap:	MaxPayload 512 bytes, PhantFunc 0
			ExtTag- RBE+
		DevCtl:	Report errors: Correctable+ Non-Fatal+ Fatal+ Unsupported+
			RlxdOrd+ ExtTag- PhantFunc- AuxPwr+ NoSnoop+
			MaxPayload 128 bytes, MaxReadReq 512 bytes
		DevSta:	CorrErr- UncorrErr- FatalErr- UnsuppReq- AuxPwr- TransPend-
		LnkCap:	Port #0, Speed 5GT/s, Width x1, ASPM L0s L1, Exit Latency L0s <2us, L1 <4us
			ClockPM+ Surprise- LLActRep- BwNot+ ASPMOptComp+
		LnkCtl:	ASPM Disabled; RCB 64 bytes Disabled- CommClk-
			ExtSynch- ClockPM- AutWidDis- BWInt- AutBWInt-
		LnkSta:	Speed 2.5GT/s, Width x1, TrErr- Train- SlotClk+ DLActive- BWMgmt- ABWMgmt-
		RootCtl: ErrCorrectable- ErrNon-Fatal- ErrFatal- PMEIntEna+ CRSVisible+
		RootCap: CRSVisible+
		RootSta: PME ReqID 0000, PMEStatus- PMEPending-
		DevCap2: Completion Timeout: Range ABCD, TimeoutDis+, LTR+, OBFF Via WAKE# ARIFwd-
		DevCtl2: Completion Timeout: 50us to 50ms, TimeoutDis-, LTR-, OBFF Disabled ARIFwd-
		LnkCtl2: Target Link Speed: 5GT/s, EnterCompliance- SpeedDis-
			 Transmit Margin: Normal Operating Range, EnterModifiedCompliance- ComplianceSOS-
			 Compliance De-emphasis: -6dB
		LnkSta2: Current De-emphasis Level: -3.5dB, EqualizationComplete-, EqualizationPhase1-
			 EqualizationPhase2-, EqualizationPhase3-, LinkEqualizationRequest-
	Capabilities: [100 v1] Advanced Error Reporting
		UESta:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UEMsk:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UESvrt:	DLP+ SDES+ TLP- FCP+ CmpltTO- CmpltAbrt- UnxCmplt- RxOF+ MalfTLP+ ECRC- UnsupReq- ACSViol-
		CESta:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- NonFatalErr-
		CEMsk:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- NonFatalErr+
		AERCap:	First Error Pointer: 00, GenCap- CGenEn- ChkCap- ChkEn-
	Capabilities: [180 v1] Vendor Specific Information: ID=0000 Rev=0 Len=028 <?>
	Capabilities: [240 v1] L1 PM Substates
		L1SubCap: PCI-PM_L1.2+ PCI-PM_L1.1+ ASPM_L1.2+ ASPM_L1.1+ L1_PM_Substates+
			  PortCommonModeRestoreTime=8us PortTPowerOnTime=10us
		L1SubCtl1: PCI-PM_L1.2- PCI-PM_L1.1- ASPM_L1.2- ASPM_L1.1-
			   T_CommonMode=1us LTR1.2_Threshold=0ns
		L1SubCtl2: T_PwrOn=10us
	Kernel driver in use: pcieport

01:00.0 Serial controller: Device 1c00:3050 (rev 10) (prog-if 05 [16850])
	Subsystem: Device 1c00:3050
	Control: I/O- Mem+ BusMaster- SpecCycle- MemWINV- VGASnoop- ParErr+ Stepping- SERR+ FastB2B- DisINTx-
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Interrupt: pin A routed to IRQ 55
	Region 0: I/O ports at <unassigned> [disabled]
	Region 1: Memory at 600000000 (32-bit, prefetchable) [size=32K]
	Region 2: I/O ports at <unassigned> [disabled]
	[virtual] Expansion ROM at 600008000 [disabled] [size=32K]
	Capabilities: [60] Power Management version 3
		Flags: PMEClk- DSI- D1- D2- AuxCurrent=375mA PME(D0+,D1-,D2-,D3hot+,D3cold+)
		Status: D0 NoSoftRst- PME-Enable- DSel=0 DScale=0 PME-
	Capabilities: [68] MSI: Enable- Count=1/32 Maskable+ 64bit+
[... 117 lines cut ...]
[ 6411.131429]  r7:ffffffff r6:bf2256c4 r5:bf225cf0 r4:bf2270a4
[ 6411.131457] [<c07951e8>] (driver_register) from [<c06f2e3c>] (__pci_register_driver+0x58/0x5c)
[ 6411.131484]  r5:bf225cf0 r4:00000000
[ 6411.131527] [<c06f2de4>] (__pci_register_driver) from [<bf22b410>] (parport_pc_init+0x310/0xf00 [parport_pc])
[ 6411.131563]  r5:bf22506c r4:00000000
[ 6411.131604] [<bf22b100>] (parport_pc_init [parport_pc]) from [<c02030c4>] (do_one_initcall+0x50/0x25c)
[ 6411.131652]  r10:c1204fc8 r9:d82a3f30 r8:bf227140 r7:00000000 r6:d843d240 r5:bf22b100
[ 6411.131680]  r4:c1204fc8
[ 6411.131707] [<c0203074>] (do_one_initcall) from [<c02c26ac>] (do_init_module+0x70/0x264)
[ 6411.131736]  r8:bf227140 r7:d843d280 r6:d843d240 r5:d843d340 r4:bf227140
[ 6411.131764] [<c02c263c>] (do_init_module) from [<c02c4d04>] (load_module+0x2368/0x266c)
[ 6411.131790]  r6:d843d240 r5:00000002 r4:00000002
[ 6411.131816] [<c02c299c>] (load_module) from [<c02c526c>] (sys_finit_module+0xd4/0xec)
[ 6411.131845]  r10:0000017b r9:d82a2000 r8:c02011c4 r7:00000003 r6:0029f150 r5:00000000
[ 6411.131870]  r4:c1204fc8
[... cut ...]
[ 6411.132021]  r7:0000017b r6:0003fce8 r5:0002abf4 r4:4c3a3e00
[ 6411.132046] Code: e5903000 e2833001 e7f33053 e2433612 (e5d33000) 
[ 6411.132075] ---[ end trace 07eba9e3068abd4c ]---
```
Kernel source acquired with rpi-source from https://github.com/RPi-Distro/rpi-source, -> "make menuconfig" - selected parallel port drivers -> "make prepare" -> "make -C . M=drivers/parport" -> "sudo insmod drivers/parport/parport.ko" -> "sudo insmod drivers/parport/parport_pc.ko irq=none" (reference: https://www.kernel.org/doc/html/latest/admin-guide/parport.html)

herrkuhn, 2021-01-02:
Research so far indicates that the problem is the BAR allocation from incompatible adresses:

```
pi@raspberrypi:~/linux $ dmesg|grep BAR
[    0.965113] pci 0000:00:00.0: BAR 8: assigned [mem 0x600000000-0x6000fffff]
[    0.965152] pci 0000:01:00.0: BAR 1: assigned [mem 0x600000000-0x600007fff pref]
[    0.965197] pci 0000:01:00.0: BAR 6: assigned [mem 0x600008000-0x60000ffff pref]
[    0.965233] pci 0000:01:00.0: BAR 0: no space for [io  size 0x0100]
[    0.965263] pci 0000:01:00.0: BAR 0: failed to assign [io  size 0x0100]
[    0.965293] pci 0000:01:00.0: BAR 2: no space for [io  size 0x0004]
[    0.965322] pci 0000:01:00.0: BAR 2: failed to assign [io  size 0x0004]
```

garlick, 2021-01-02:
Hmm, I hope we're not running up against this "IO BAR" issue, but I suspect that might be it:

https://www.raspberrypi.org/forums/viewtopic.php?t=288902

Specifically, see this interchange:

>>> PhilE wrote: 
>>> Thu Oct 22, 2020 3:23 pm
>>> I'm looking into the BAR 5 problem - it's possibly down to it being an IO BAR rather than a memory BAR.
>>
>> Does it even support IO BARs? It'd need to do it via MMIO as I/O ports are a weird legacy thing from x86. VGA had some I/O ports so it is probably for that.
>>
> Short answer - no. From the datasheet:
> "Supports accessing external PCIe configuration space and memory space (no support for I/O space)."

garlick, 2021-01-02:
To get another data point, I'm setting up a test with

https://media.startech.com/cms/pdfs/pex1p_datasheet.pdf

And latest (5.10) kernel from

https://github.com/raspberrypi/linux/

Kernel building now...

garlick, 2021-01-03:
OK, I'm getting the same oops as you.  Here's the lspci -vvv for my card:
```
      Subsystem: Oxford Semiconductor Ltd Device c110
        Control: I/O- Mem- BusMaster- SpecCycle- MemWINV- VGASnoop- ParErr+ Stepping- SERR+ FastB2B- DisINTx-
        Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
        Interrupt: pin A routed to IRQ 64
        Region 0: I/O ports at <unassigned> [disabled]
        Region 1: I/O ports at <unassigned> [disabled]
        Capabilities: [40] Power Management version 3
                Flags: PMEClk- DSI- D1+ D2+ AuxCurrent=55mA PME(D0-,D1+,D2+,D3hot+,D3cold+)
                Status: D0 NoSoftRst+ PME-Enable- DSel=0 DScale=0 PME-
        Capabilities: [50] MSI: Enable- Count=1/1 Maskable- 64bit+
                Address: 0000000000000000  Data: 0000
        Capabilities: [70] Express (v1) Legacy Endpoint, MSI 00
                DevCap: MaxPayload 128 bytes, PhantFunc 0, Latency L0s <128ns, L1 <2us
                        ExtTag- AttnBtn- AttnInd- PwrInd- RBE+ FLReset-
                DevCtl: Report errors: Correctable- Non-Fatal- Fatal- Unsupported-
                        RlxdOrd+ ExtTag- PhantFunc- AuxPwr- NoSnoop-
                        MaxPayload 128 bytes, MaxReadReq 512 bytes
                DevSta: CorrErr- UncorrErr- FatalErr- UnsuppReq- AuxPwr+ TransPend-
                LnkCap: Port #0, Speed 2.5GT/s, Width x1, ASPM L0s L1, Exit Latency L0s <512ns, L1 <64us
                        ClockPM+ Surprise- LLActRep- BwNot- ASPMOptComp-
                LnkCtl: ASPM Disabled; RCB 64 bytes Disabled- CommClk-
                        ExtSynch- ClockPM- AutWidDis- BWInt- AutBWInt-
                LnkSta: Speed 2.5GT/s, Width x1, TrErr- Train- SlotClk+ DLActive- BWMgmt- ABWMgmt-
        Capabilities: [100 v1] Device Serial Number 00-30-e0-11-11-00-01-10
        Capabilities: [110 v1] Power Budgeting <?>
        Kernel driver in use: parport_pc
        Kernel modules: parport_pc
```
and the oops (just ran modprobe parport_pc with no module options):
```
[  213.224068] PCI parallel port detected: 1415:c110, I/O at 0x0(0x0), IRQ 64
[  213.224259] 8<--- cut here ---
[  213.226858] Unable to handle kernel paging request at virtual address fee00001
[  213.229433] pgd = a425deda
[  213.231830] [fee00001] *pgd=80000000007003, *pmd=00000000
[  213.234439] Internal error: Oops: 206 [#1] SMP ARM
[  213.236830] Modules linked in: parport_pc(+) parport binfmt_misc sha256_generic cfg80211 rfkill 8021q garp stp llc v3d gpu_sched raspberrypi_hwmon dwc2 roles vc4 cec bcm2835_codec(C) drm_kms_helper bcm2835_v4l2(C) bcm2835_isp(C) v4l2_mem2mem bcm2835_mmal_vchiq(C) videobuf2_vmalloc videobuf2_dma_contig videobuf2_memops videobuf2_v4l2 videobuf2_common vc_sm_cma(C) drm snd_bcm2835(C) drm_panel_orientation_quirks videodev mc snd_soc_core snd_compress snd_pcm_dmaengine snd_pcm rpivid_mem snd_timer snd syscopyarea sysfillrect sysimgblt fb_sys_fops backlight uio_pdrv_genirq uio ip_tables x_tables ipv6
[  213.245275] CPU: 0 PID: 516 Comm: modprobe Tainted: G         C        5.10.4-v7l-parport+ #5
[  213.248072] Hardware name: BCM2711
[  213.250894] PC is at clear_epp_timeout+0x24/0xc4 [parport_pc]
[  213.253726] LR is at parport_EPP_supported+0x1c/0x78 [parport_pc]
[  213.256543] pc : [<bf2b9b2c>]    lr : [<bf2b9ff8>]    psr: a0000013
[  213.259373] sp : c3e5bae0  ip : c3e5baf0  fp : c3e5baec
[  213.262199] r10: c22e9700  r9 : 00000000  r8 : c2fdd428
[  213.265022] r7 : c1816070  r6 : c2fdd400  r5 : c1205048  r4 : c3195c00
[  213.267868] r3 : fee00001  r2 : 5b9855aa  r1 : c22e9700  r0 : c3195c00
[  213.270717] Flags: NzCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
[  213.273602] Control: 30c5383d  Table: 032ac700  DAC: fffffffd
[  213.276490] Process modprobe (pid: 516, stack limit = 0x30cf4815)
[  213.279385] Stack: (0xc3e5bae0 to 0xc3e5c000)
[  213.282285] bae0: c3e5bb04 c3e5baf0 bf2b9ff8 bf2b9b14 c3195c00 c1205048 c3e5bba4 c3e5bb08
[  213.285245] bb00: bf2ba780 bf2b9fe8 00000005 00000000 c35eb800 00000000 00000040 00000000
[  213.288238] bb20: c2fdd428 00000000 00000000 00000000 c35a9a80 00000000 c3e5bb74 c3e5bb48
[  213.291242] bb40: c028513c c02842cc c1801e40 c041f9bc c3e5bba4 c3e5bb60 c1205048 00000000
[  213.294262] bb60: c1816000 00000000 c3e5bb94 c3e5bb78 c0b59ca8 5b9855aa c3e5bb9c bf2bda44
[  213.297292] bb80: 00000000 c1816000 00000000 bf2bd510 00000040 00000000 c3e5bbfc c3e5bba8
[  213.300337] bba0: bf2bb858 bf2ba410 c1816070 00000080 bf2bf0a0 00000000 bf2be178 bf2be1b4
[  213.303402] bbc0: c1816070 c2fddc80 bf2bd468 00000000 c3e5bbfc c1816070 c1816000 bf2bf070
[  213.306502] bbe0: 00000000 bf2bf0a0 bf2bda44 00000000 c3e5bc24 c3e5bc00 c07a33cc bf2bb6bc
[  213.309603] bc00: c1816070 c13fd62c c13fd634 00000000 bf2bf0a0 00000023 c3e5bc54 c3e5bc28
[  213.312702] bc20: c08405f4 c07a3328 00000000 c1816070 bf2bf0a0 bf2bf0a0 c0840c08 00000000
[  213.315806] bc40: fffffffd bf2bf3f8 c3e5bc6c c3e5bc58 c08409b8 c0840410 c1816070 00000000
[  213.318922] bc60: c3e5bc8c c3e5bc70 c0840c00 c0840958 00000000 bf2bf0a0 c1816070 c0840c08
[  213.322051] bc80: c3e5bcac c3e5bc90 c0840c70 c0840ba4 c20779b4 c1205048 bf2bf0a0 c0840c08
[  213.325168] bca0: c3e5bcdc c3e5bcb0 c083e56c c0840c14 c3e5bce8 c1a29658 c20779b4 5b9855aa
[  213.328279] bcc0: bf2bf0a0 c12e7c64 c35a9b80 00000000 c3e5bcec c3e5bce0 c083fe80 c083e4f4
[  213.331407] bce0: c3e5bd14 c3e5bcf0 c083f7d0 c083fe60 bf2bde0c c3e5bd00 bf2bf0a0 bf2bde0c
[... cut ...]
[  213.557801] [<c02ccebc>] (sys_finit_module) from [<c0200040>] (ret_fast_syscall+0x0/0x28)
[  213.561613] Exception stack(0xc3e5bfa8 to 0xc3e5bff0)
[  213.565403] bfa0:                   664b8b00 00000000 00000004 0002d064 00000000 0002ec3c
[  213.569238] bfc0: 664b8b00 00000000 00000000 0000017b 006ebe50 00000000 006ebda8 00000000
[  213.573062] bfe0: bead82b8 bead82a8 00022cb8 b6c25af0
[  213.576865]  r7:0000017b r6:00000000 r5:00000000 r4:664b8b00
[  213.580655] Code: e5903000 e2833001 e7f33053 e2433612 (e5d33000) 
[  213.584444] ---[ end trace 792fd8b17530f5f3 ]---
```

garlick, 2021-01-03:
should have noted this too:
```
$ dmesg|grep BAR
[    1.282152] pci 0000:01:00.0: BAR 0: no space for [io  size 0x0008]
[    1.284620] pci 0000:01:00.0: BAR 0: failed to assign [io  size 0x0008]
[    1.287138] pci 0000:01:00.0: BAR 1: no space for [io  size 0x0004]
[    1.289612] pci 0000:01:00.0: BAR 1: failed to assign [io  size 0x0004]
```

herrkuhn, 2021-01-03:
> [    1.287138] pci 0000:01:00.0: BAR 1: no space for [io  size 0x0004]
> [    1.289612] pci 0000:01:00.0: BAR 1: failed to assign [io  size 0x0004]

Same as mine, the io size of 0x0004 leads me to speculate that's the address for 0x378, since that would be enough for data/status/control registers.

garlick, 2021-01-03:
Same model card in a  PC:
```
[    6.742514] parport0: PC-style at 0xdc00 (0xd880), irq 16, using FIFO [PCSPP,TRISTATE,COMPAT,EPP,ECP]
```
And `lspci -vvv` 
```
03:00.0 Parallel controller: Oxford Semiconductor Ltd Device c110 (prog-if 02 [ECP])
        Subsystem: Oxford Semiconductor Ltd Device c110
        Physical Slot: 0
        Control: I/O+ Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
        Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
        Latency: 0, Cache Line Size: 32 bytes
        Interrupt: pin A routed to IRQ 16
        Region 0: I/O ports at dc00 [size=8]
        Region 1: I/O ports at d880 [size=4]
        Capabilities: [40] Power Management version 3
                Flags: PMEClk- DSI- D1+ D2+ AuxCurrent=55mA PME(D0-,D1+,D2+,D3hot+,D3cold+)
                Status: D0 NoSoftRst+ PME-Enable- DSel=0 DScale=0 PME-
        Capabilities: [50] MSI: Enable- Count=1/1 Maskable- 64bit+
                Address: 0000000000000000  Data: 0000
        Capabilities: [70] Express (v1) Legacy Endpoint, MSI 00
                DevCap: MaxPayload 128 bytes, PhantFunc 0, Latency L0s <128ns, L1 <2us
                        ExtTag- AttnBtn- AttnInd- PwrInd- RBE+ FLReset-
                DevCtl: Report errors: Correctable- Non-Fatal- Fatal- Unsupported-
                        RlxdOrd+ ExtTag- PhantFunc- AuxPwr- NoSnoop-
                        MaxPayload 128 bytes, MaxReadReq 512 bytes
                DevSta: CorrErr- UncorrErr- FatalErr- UnsuppReq- AuxPwr+ TransPend-
                LnkCap: Port #0, Speed 2.5GT/s, Width x1, ASPM L0s L1, Exit Latency L0s unlimited, L1 unlimited
                        ClockPM+ Surprise- LLActRep- BwNot- ASPMOptComp-
                LnkCtl: ASPM Disabled; RCB 64 bytes Disabled- CommClk+
                        ExtSynch- ClockPM- AutWidDis- BWInt- AutBWInt-
                LnkSta: Speed 2.5GT/s, Width x1, TrErr- Train- SlotClk+ DLActive- BWMgmt- ABWMgmt-
        Capabilities: [100 v1] Device Serial Number 00-30-e0-11-11-00-01-10
        Capabilities: [110 v1] Power Budgeting <?>
        Kernel driver in use: parport_pc
        Kernel modules: parport_pc
```

According to comment at the top of `parport_pc.c`, the two segments are 1) the data/status/control registers, and 2) the optional EPP/ECP registers.  The datasheet for the chip in my card describes just the two I/O mapped BARs for the parallel port:

https://www.semiconductorstore.com/pdf/newsite/oxford/OXPCIe952_ds.pdf

If the Broadcom 2711's PCI bridge doesn't support mapping I/O addresses, and the cards don't offer a memory mapped alternative, I think we may be at a dead end.