worlickwerx/pi-parport #51: Use DIP switch and remove R1 DNP pads in future board revs

quorten, 2021-01-02:
Thinking through it more, it's probably better to just use a DIP switch and remove the R1 pad from future board revisions.  Removable jumpers feels clumsy.  The cost savings of not using a switch don't seem to make sense for us, and a rugged write-enable connection is probably not important for our board.  The motivation of the Pi HAT template including both may have been to simply make it easy for downstream designers to choose one way or the other, but not both.

A future board revision can incorporate these changes.

garlick, 2021-01-02:
What would be the downside to permanently enabling EEPROM writes?  This isn't really a plug and play project, so it seems like a low magnitude foot-gun.  Unless maybe a spurious write is an actual issue and leaving it writable would reduce its reliability.

quorten, 2021-01-02:
Good point, I can't think of anything that could really much go wrong with spurious writes given that I2C0 is generally considered to be reserved.

garlick, 2024-08-29:
In v5 I used the "solder bridge" (open) footprint that came with the HAT template.  Maybe OK?

quorten, 2024-09-01:
Yes, and actually I've been thinking about another technique that works well on the v4 board.  I have a SOIC clip that I can attach to the EEPROM chip.  From there, it is easy to wire write-enable.  Or, alternatively, I could clip it to the unpopulated pads on the board.  I think that approach works nicely for v4 board, so I could make a documentation update to that respect.

The v5 board uses the reverse logic on write-protect, so this idea is not applicable there.