worlickwerx/pi-parport #56: Rpi 3 kernel 5.4.51 errors while make

cervellopl, 2021-03-07:
Hi,

I getting errors while build drivers. Details below.
Best regards, Maciej

make -C /lib/modules/5.4.51-v7+/build M=/home/astroberry/dr2/pi-parport/driver modules
make[1]: Entering directory '/usr/src/linux-headers-5.4.51-v7+'
  CC [M]  /home/astroberry/dr2/pi-parport/driver/parport/share.o
  CC [M]  /home/astroberry/dr2/pi-parport/driver/parport/ieee1284.o
  CC [M]  /home/astroberry/dr2/pi-parport/driver/parport/ieee1284_ops.o
  CC [M]  /home/astroberry/dr2/pi-parport/driver/parport/procfs.o
  CC [M]  /home/astroberry/dr2/pi-parport/driver/parport/daisy.o
  CC [M]  /home/astroberry/dr2/pi-parport/driver/parport/probe.o
  LD [M]  /home/astroberry/dr2/pi-parport/driver/parport/parport.o
  CC [M]  /home/astroberry/dr2/pi-parport/driver/parport/lp.o
  CC [M]  /home/astroberry/dr2/pi-parport/driver/parport/ppdev.o
  CC [M]  /home/astroberry/dr2/pi-parport/driver/parport_gpio.o
/home/astroberry/dr2/pi-parport/driver/parport_gpio.c: In function 'parport_gpio_write_data':
/home/astroberry/dr2/pi-parport/driver/parport_gpio.c:71:60: error: passing argument 3 of 'gpiod_set_array_value' from incompatible pointer type [-Werror=incompatible-pointer-types]
  gpiod_set_array_value(ctx->data->ndescs, ctx->data->desc, val);
                                                            ^~~
In file included from /home/astroberry/dr2/pi-parport/driver/parport_gpio.c:20:
./include/linux/gpio/consumer.h:124:25: note: expected 'struct gpio_array *' but argument is of type 'int *'
      struct gpio_array *array_info,
      ~~~~~~~~~~~~~~~~~~~^~~~~~~~~~
/home/astroberry/dr2/pi-parport/driver/parport_gpio.c:71:2: error: too few arguments to function 'gpiod_set_array_value'
  gpiod_set_array_value(ctx->data->ndescs, ctx->data->desc, val);
  ^~~~~~~~~~~~~~~~~~~~~
In file included from /home/astroberry/dr2/pi-parport/driver/parport_gpio.c:20:
./include/linux/gpio/consumer.h:122:5: note: declared here
 int gpiod_set_array_value(unsigned int array_size,
     ^~~~~~~~~~~~~~~~~~~~~
/home/astroberry/dr2/pi-parport/driver/parport_gpio.c: In function 'parport_gpio_write_control':
/home/astroberry/dr2/pi-parport/driver/parport_gpio.c:106:66: error: passing argument 3 of 'gpiod_set_array_value' from incompatible pointer type [-Werror=incompatible-pointer-types]
  gpiod_set_array_value(ctx->control->ndescs, ctx->control->desc, value);
                                                                  ^~~~~
In file included from /home/astroberry/dr2/pi-parport/driver/parport_gpio.c:20:
./include/linux/gpio/consumer.h:124:25: note: expected 'struct gpio_array *' but argument is of type 'int *'
      struct gpio_array *array_info,
      ~~~~~~~~~~~~~~~~~~~^~~~~~~~~~
/home/astroberry/dr2/pi-parport/driver/parport_gpio.c:106:2: error: too few arguments to function 'gpiod_set_array_value'
  gpiod_set_array_value(ctx->control->ndescs, ctx->control->desc, value);
  ^~~~~~~~~~~~~~~~~~~~~
In file included from /home/astroberry/dr2/pi-parport/driver/parport_gpio.c:20:
./include/linux/gpio/consumer.h:122:5: note: declared here
 int gpiod_set_array_value(unsigned int array_size,
     ^~~~~~~~~~~~~~~~~~~~~
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:266: /home/astroberry/dr2/pi-parport/driver/parport_gpio.o] Error 1
make[1]: *** [Makefile:1709: /home/astroberry/dr2/pi-parport/driver] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.4.51-v7+'
make: *** [Makefile:12: modules] Error 2

garlick, 2021-03-08:
Hi, one thing to try is to run the `get_parport_src.sh`  script down in the drivers directory.  This will update the parport sources to match your installed kernel (assuming astroberry uses same kernel package names as rasbpian).  I'm not sure if that's going to fix the `parport_gpio.c` errors though.  Try?

If it doesn't work I'll go grab that specific kernel and see if I can make a quick patch for you to apply.

cervellopl, 2021-03-08:
Thanks for tip. After running script I build driver without issues

closed 2021-03-08 by cervellopl