worlickwerx/pi-parport #57: probe of ppgpio@0 failed with error -22

cervellopl, 2021-03-08:
After boot and loading modules as in instruction I had something like this:

Mar  8 18:45:29 astroberry kernel: [ 3497.542536] parport: loading out-of-tree module taints kernel.
Mar  8 18:45:36 astroberry kernel: [ 3504.134802] pinctrl-bcm2835 3f200000.gpio: pin gpio11 already requested by 3f204000.spi; cannot claim for ppgpio@0
Mar  8 18:45:36 astroberry kernel: [ 3504.134833] pinctrl-bcm2835 3f200000.gpio: pin-11 (ppgpio@0) status -22
Mar  8 18:45:36 astroberry kernel: [ 3504.134848] pinctrl-bcm2835 3f200000.gpio: could not request pin 11 (gpio11) from group gpio11  on device pinctrl-bcm2835
Mar  8 18:45:36 astroberry kernel: [ 3504.134859] parport-gpio ppgpio@0: Error applying setting, reverse things back
Mar  8 18:45:36 astroberry kernel: [ 3504.134963] parport-gpio: probe of ppgpio@0 failed with error -22
Mar  8 18:45:56 astroberry kernel: [ 3524.533219] ppdev: user-space parallel port driver

Any suggestions?
Best regards,
Maciej

garlick, 2021-03-08:
It sounds like you'll need to disable `spi0` (pin 11 is assigned to spi0 SCLK per [this doc](https://www.raspberrypi.org/documentation/hardware/raspberrypi/spi/README.md))
Commenting out this in your `/boot/config.txt` and rebooting may do it

```
dtparam=spi=on
```
Or if another device is using spi0 you might need to comment that out.

cervellopl, 2021-03-08:
Done but still have:
Mar  8 20:03:55 astroberry kernel: [  499.281789] parport: loading out-of-tree module taints kernel.
Mar  8 20:04:03 astroberry kernel: [  507.668429] pinctrl-bcm2835 3f200000.gpio: pin gpio2 already requested by 3f804000.i2c; cannot claim for ppgpio@0
Mar  8 20:04:03 astroberry kernel: [  507.668450] pinctrl-bcm2835 3f200000.gpio: pin-2 (ppgpio@0) status -22
Mar  8 20:04:03 astroberry kernel: [  507.668464] pinctrl-bcm2835 3f200000.gpio: could not request pin 2 (gpio2) from group gpio2  on device pinctrl-bcm2835
Mar  8 20:04:03 astroberry kernel: [  507.668474] parport-gpio ppgpio@0: Error applying setting, reverse things back
Mar  8 20:04:03 astroberry kernel: [  507.668569] parport-gpio: probe of ppgpio@0 failed with error -22

garlick, 2021-03-08:
Looks like a conflict with i2c1.   You could try commenting out
```
dtparam=i2c_arm=on
```

cervellopl, 2021-03-08:
Ok it looks after that parport correct running driver. But how about EEPROM? I2C is not available after this operation.

garlick, 2021-03-08:
I think the fact that it is not available from linux does not prevent it from being accessed during boot.

For flashing, which involves temporarily enabling i2c, I _think_ you should be able to follow the instructions [here](https://github.com/worlickwerx/pi-parport/tree/master/eeprom/README.md), even with `dtparam=i2c_arm=on` commented out.  The `parport_gpio` driver does not need to be loading during this process (although as I recall, there is no conflict reported).

quorten, 2021-03-09:
Yeah, the flashing instructions point you to using i2c0 rather than i2c1.  However, it may be easier to use the GPIO-based i2c driver instead, that can be dynamically loaded without reboot and is apparently officially recommended in preference to using i2c0.  I haven't tested it yet, though, but you would do it like this:

```
# Dynamically install the overlay and driver.
sudo dtoverlay i2c-gpio i2c_gpio_sda=0 i2c_gpio_scl=1 bus=3

# Run the flashing commands here.

# Remove the overlay.
sudo dtoverlay -r i2c-gpio
# N.B.: Any auto-loaded kernel modules will remain resident until explicitly removed.
```

quorten, 2021-03-09:
As a side note, I'm curious to see how pi-parport fits into the astroberry picture.  Mentioning the i2c conflict, it's in the list of the number of things we'd like to change in a future revision, along with making other adjustments to ease the simultaneous connection of an RTC, GPS, power button, etc.