worlickwerx/pi-parport #58: Advise for possible use for Diamond Rio MP3 player

deanhuff, 2021-05-19:
I have been investigating using GPIO pins to a DB25 breakout board along with SMI (secondary memory interface) to implement a parallel port so that I can upload/download music to my MP3 player from 1998.  Source code to the mp3 software is still maintained in debian rio package here: https://sources.debian.org/src/rio/1.07-14/  pertinent info is in rio.txt.

do you think a pi-parport would be a good option?

garlick, 2021-05-19:
A quick scan of the code shows it uses `outb()`  / `inb()` to access the parallel port directly from user space.

That doesn't work with pi-parport - it would need to be converted to use ppdev ioctls, and then it may or may not work. ppdev adds latency compared to memory mapped I/O but that is mitigated somewhat by the fact that the pi may be faster at some things than a PC of 1998.

deanhuff, 2021-05-20:
understood, thanks for the reply.

closed 2021-05-20 by deanhuff