worlickwerx/pi-parport #64: add a Qwiic connector

garlick, 2024-08-19:
I'm not sure if we have already used all the I2C GPIOs but if not, it could be useful in some applications have have a  [Quiic](https://www.sparkfun.com/Qwiic) connector available for expansion.

quorten, 2024-08-24:
I looked at the board and it should be fairly straightforward to free up the I2C GPIOs.  Some other useful pins we already have free, a few others should probably be freed up.  Here are my notes:

* BCM2 (pin 3) and BCM3 (pin 5) should be freed up to allow for use of I2C1 in ALT0 function selection.  BCM5 (pin 29) and BCM7 (pin 26) can be used instead for pi-parport.

* BCM14 (pin 8) and BCM15 (pin 10) should be kept free to allow for use of UART0 in ALT0 function selection.

* BCM16 (pin 36) should be kept free, BCM17 (pin 11) should be freed up to allow for use of RTS-CTS hardware handshaking on UART0 in ALT3 function selection.  BCM27 (pin 13) can be used instead for pi-parport.

* BCM12 (pin 32) should be kept free to allow for use of PWM0 in ALT0 function selection.

garlick, 2024-08-24:
Nice analysis, thanks!

I suppose we can distribute two DT overlays, one for the older boards and one for the newer and select them with the EEPROM.